NASA-JSC-Robotics/valkyrie #5: Ankle roll torques are communicated to IHMC controller.

SylvainBertrand, 2015-11-19:


jlack1987, 2015-11-19:
We already fixed this and send you the updated debians in an email this morning. Follow the instructions in the email to install them.

SylvainBertrand, 2015-11-19:
Alright I’ll close the ticket next time I run Val. Do you know how to add a “Review” column in the zenhub board (like in Jira)?

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

kodyensley, 2015-11-20:
I can add/modify the board. What would you like to add/remove or rename?
-Kody

https://www.avast.com/antivirus

SylvainBertrand, 2015-11-20:
Yo!

Sorry, I just saw your email. I like the current setup, thanks!

Sylvain

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 https://www.avast.com/antivirus

closed 2015-12-03 by jlack1987