NASA-JSC-Robotics/valkyrie #9: shared_memory_transport_sliderboard doesn't get cloned when cloning a workspace from the workspace files

jlack1987, 2016-01-17:
Recreating this ticket from @scpeters , trying to clone a workspace from public_full_workspace.yaml gives following error:

=== ./shared_memory_transport_sliderboard (git) ===
Could not clone repository 'git@github.com:NASA-JSC-Robotics/shared_memory_transport_sliderboard.git': Cloning into '.'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

pstrawse, 2016-01-18:
There was a permissions issue.  The permissions for this repo has been updated and this issue should be fixed.

jlack1987, 2016-01-18:
Tested just now and can confirm the sliderboard is clone successfully.

closed 2016-01-18 by jlack1987