NASA-JSC-Robotics/valkyrie #11: Trouble running feature/ModifiedControllerTest

scpeters, 2016-01-29:
I've been trying to test the `feature/ModifiedControllerTest` branches in

* val_controllers
* val_deploy
* val_gazebo

When I try to launch `val_gazebo val_sim_pinned.launch`, I see the following error that makes me think a controller is not loading properly:

~~~
process[rttlua_t2_12431_2450744123138483789-1]: started with pid [12452]
Real-time memory: 517904 bytes free of 524288 allocated.
OROCOS RTTLua 1.0-beta5 / Lua 5.1.5 (gnulinux)
/opt/ros/indigo/bin/rttlua-gnulinux: ...kyrie/src/val_deploy/scripts/startup_val_control.lua:18: attempt to call method 'import' (a nil value)
stack traceback:
	...kyrie/src/val_deploy/scripts/startup_val_control.lua:18: in main chunk
	[C]: in function 'require'
	...ws/valkyrie/src/val_deploy/scripts/startupRemote.lua:44: in main chunk
	[C]: ?
TLSF bytes allocated=524288 overhead=6384 max-used=6384 currently-used=6384 still-allocated=0
[rttlua_t2_12431_2450744123138483789-1] process has finished cleanly
~~~

@hsu @MathisFrankB

scpeters, 2016-01-29:
This link might be relevant, but I have no prior experience with lua, so I'm at a loss:

* http://lists.mech.kuleuven.be/pipermail/orocos-users/2013-July/007328.html

jlack1987, 2016-01-29:
@dustingooding is our in house orocos/lua expert, and a quick look by him leads us to believe you might be missing a .rttlua file in your home directory. Can you verify that you have the file ~/.rttlua and that its contents are something like the following:

require("rttlib")
rttlib.color=true
require("complete")

scpeters, 2016-01-29:
I definitely don't have that file in my home folder. I'll try creating it. If this is required, we should document it somewhere.

scpeters, 2016-01-29:
Ok, that helped; I'm getting a little farther. Now I see:

~~~
/opt/ros/indigo/bin/rttlua-gnulinux: ...ws/valkyrie/src/val_deploy/scripts/startupRemote.lua:54: module 'posix' not found:
	no field package.preload['posix']
	no file '/data_fast/scpeters/ws/valkyrie/src/val_deploy/scripts/posix.lua'
	no file '/data_fast/scpeters/ws/valkyrie/src/rttlua_completion/posix.lua'
	no file './posix.lua'
	no file '/usr/local/share/lua/5.1/posix.lua'
	no file '/usr/local/share/lua/5.1/posix/init.lua'
	no file '/usr/local/lib/lua/5.1/posix.lua'
	no file '/usr/local/lib/lua/5.1/posix/init.lua'
	no file '/usr/share/lua/5.1/posix.lua'
	no file '/usr/share/lua/5.1/posix/init.lua'
	no file '/opt/ros/indigo/share/lua/5.1/posix.lua'
	no file '/data_fast/scpeters/ws/valkyrie/devel/lib/posix.so'
	no file '/data_fast/scpeters/ws/valkyrie/src/rttlua_completion/posix.so'
	no file './posix.so'
	no file '/usr/local/lib/lua/5.1/posix.so'
	no file '/usr/lib/x86_64-linux-gnu/lua/5.1/posix.so'
	no file '/usr/lib/lua/5.1/posix.so'
	no file '/usr/local/lib/lua/5.1/loadall.so'
	no file '/opt/ros/indigo/lib/lua/5.1/posix.so'
stack traceback:
	[C]: in function 'require'
	...ws/valkyrie/src/val_deploy/scripts/startupRemote.lua:54: in main chunk
	[C]: ?
TLSF bytes allocated=524288 overhead=6384 max-used=6384 currently-used=6384 still-allocated=0
~~~

I see the following command recommended in a [comment in startupRemote.lua](https://github.com/NASA-JSC-Robotics/val_deploy/blob/develop/scripts/startupRemote.lua#L46-L53):

~~~
$ sudo luarocks install luaposix
~~~

I'll try that now. Again, I recommend that the documentation include these steps.

jlack1987, 2016-01-29:
It is documented [here](https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Get-Valkyrie-Code). If you click the link to [Setup Valkyrie Development Environment](https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Setup-Valkyrie-Developer-Environment) on that page and follow the steps there it will set it all up for you and create that file

scpeters, 2016-01-29:
Good point. I didn't follow all those steps. I see that the `.rttlua` is part of the `nasa-indigo-workspace_1.0.0_amd64.deb`

scpeters, 2016-01-29:
I see that the `rosdep` command would have installed `lua-posix` as well.

scpeters, 2016-01-29:
It works if you follow the instructions.

closed 2016-01-29 by scpeters

kodyensley, 2016-01-29:
Awesome, thanks for testing those out!

dustingooding, 2016-01-30:
There's a PR on val_deploy that's related to this, and that I'm not
necessarily opposed to.  We'll talk about it internally, and potentially
change the instructions a bit.  But yes, thanks for validating them!