NASA-JSC-Robotics/valkyrie #12: Valkyrie simulation instability, lack of joint damping, friction

scpeters, 2016-01-30:
@hsu and I have been investigating some reported simulation instabilities of valkyrie. We are using the `feature/ModifiedControllerTest` branches of `val_controllers`, `val_deploy`, and `val_gazebo` ([workspace here](https://github.com/scpeters/val_workspaces/blob/bdbbc37e5d/val_pinned_sim.yaml)) and launching `val_gazebo val_sim_pinned.launch`.

The following links are password-protected videos. The password is `valkyrie` in each case.

* Here's a screen capture of the simulation with the [ODE physics engine and default parameters](https://vimeo.com/153575829). The torso doesn't do normal things.

* @hsu found that changing the SOR parameter from 1.3 to 1.0 [yields a substantial improvement](https://vimeo.com/153575828), and increasing iterations can help as well. I'll submit a pull request to change the SOR parameter for the `val_sim_pinned` world file.

* Also, the dart physics engine seems to [behave well with default parameters](https://vimeo.com/153575827).

@dljsjr 

One thing we noticed is that there is basically no joint damping or joint friction. We found that it was important to set these parameters for Atlas. Can you recommend any values for these parameters?

scpeters, 2016-01-30:
cc @MathisFrankB

scpeters, 2016-01-30:
change default SOR parameter and iterations in https://github.com/NASA-JSC-Robotics/val_gazebo/pull/1

dljsjr, 2016-01-30:
I can look in to if we did anything outside of the defaults for Atlas in gazebo but I don't know that we did.

jlack1987, 2016-01-30:
I suppose I don't mind adding artificial damping to the joints if it's the only way to  make the sim behave, just a bit of a pain since it's more parameters that differ between the actual robot URDF and the sim robot URDF since the actual damping/friction on the output of the real robot is actually crazy low. Also then people need to be careful to make sure they don't use the sim URDF when computing dynamics when controlling the real robot, which I know some people have been doing. 

Let me know though if you want me to pull the trigger on that, should only take me ~30 minutes to extract those parameters out and have sim and real robot use different numbers. Also it would be nice if you could give us some suggestions for what you want the parameters should be.

dljsjr, 2016-01-31:
My thought on it, which might be a little reductionist, is that that the sim model shouldn't differ from the physical robot model just to accommodate the physics. Obviously this doesn't always work, we've had to make hacks to robot models in the past to get them to work in Simulation Construction Set's physics as well, but it makes me sad face every time we have to do it.  If it's the only way to solve the problem then so be it, though.

scpeters, 2016-01-31:
If the actual robot has very little damping, that's great. The goal is to match the real behavior. Probably  what we need instead is a reference for what the real robot behavior looks like when the `feature/ModifiedControllerTest` controller is applied, since we're currently in the dark on that. It looked like the damping was low, so I just mentioned that, but we really should be trying to match some experimental data.

jlack1987, 2016-01-31:
Alright we'll see if we can get a reference on it this week for you guys.

iamwolf, 2016-02-01:
We didn't change the damping but got it to behave by increasing iterations from 100 to 500-1000. Alternatively, DART or Bullet do the job but require compiling from source - ODE has been more prone to misbehave in our tests. I'd be interested to learn how much the sim matches the real robot as well.

hsu, 2016-02-01:
I just want to reiterate Steve's comment here, the main change here that stabilizes the solver was reducing `sor` from 1.3 to 1.0. Higher values of `sor` at 1.3 was causing the simulation to be unstable. Joint viscous *damping* does not need to be changed.

On the topic of damping... joint damping does not appear to be necessary for stability for the test scenario I am running under `val_gazebo`. However, I was wondering does the real robot have any gearbox or mechanical reduction mechanism transmissions for the [high torque] actuated joints such as torso pitch? Mechanical reduction mechanisms such as gearbox/belt/harmonic gear/4 bar linkages/etc is obviously not modeled in the current model, the resistance to joint position change of such mechanisms can behave roughly similar to viscous damping, the actuators sometimes have built-in resistance to change as well (for example some pneumatic or hydraulic actuators will damp backdriving joint forces). So joint damping (or joint friction) is often not too far from real systems.

Each engine has its trade-offs. DART uses internal coordinates solver for bilateral constraints, but uses direct solver for contacts. DART does very well without contacts (pinned suspended robots), but will slow down much quicker than iterative solvers (in Bullet/ODE) or fail to converge when multiple frictional contact constraints are present (grasping or standing robots). Bullet2 **just** needs a bit of work to get the [implicit joint damping working](https://bitbucket.org/osrf/gazebo/issues/619/implement-implicit-stiffness-and-damping) to get stability and performance improvements for other robots that needs joint damping.

iamwolf, 2016-02-02:
Thanks hsu for the insightful explanation, much appreciated - I learnt a lot

jlack1987, 2016-02-02:
Our complex, pushrod actuated joints(torso/ankles) don't have gear boxes. Any "gearing" with those joints is done via a really low friction lead screw. The other simple 1-DOF joints do have gear boxes; however, our force/torque sensor is downstream of all of this stuff, so almost all of the friction and natural damping is compensated for in the actuator control, resulting in joints with extremely low friction/damping on the joint output.

MathisFrankB, 2016-02-02:
Here is a video of us running the joint position controller on the robot. We made some slight modifications to make it safe for the robot, but if you pull and updated version of the branch feature/ModifiedControllerTest you will be able to launch the exact controller used in the video in simulation:  https://youtu.be/ue51i2MOuik

scpeters, 2016-02-02:
One thing I noticed from the video is that the harness is attached to the torso, while in the simulation, the pelvis is pinned. Should we pin the torso instead of the pelvis in the simulation?

hsu, 2016-02-02:
Thanks for the great video! A picture is worth a thousand words, so at 30fps...

The changes I would like to propose here is below (to be added to the world file):

~~~
    <physics type="ode">
      <gravity>0.000000 0.000000 -9.810000</gravity>
      <ode>
        <solver>
          <type>quick</type>
          <iters>100</iters>
          <precon_iters>0</precon_iters>
          <sor>1.000000</sor>
        </solver>
        <constraints>
          <cfm>0.000000</cfm>
          <erp>0.800000</erp>
          <contact_max_correcting_vel>2000.000000</contact_max_correcting_vel>
          <contact_surface_layer>0.01000</contact_surface_layer>
        </constraints>
      </ode>
    </physics>
~~~

note that `contact_max_correcting_vel` was also set too low and truncating necessary constraint torque for joints to stay together.

[This is how I arrived at these values](https://youtu.be/j0jCCcgLyEA), once we establish more test scenarios and more precise metrics (to be written as integration tests?) then the values can be further refined?

jlack1987, 2016-02-02:
Yeah that sounds good to us

hsu, 2016-02-03:
what branch of `val_gazebo` should we create the PR towards?

jlack1987, 2016-02-03:
Make the PR to develop and we'll pull that into any feature branches that need it

scpeters, 2016-02-03:
Should we also consider pinning the torso instead of the pelvis in the example world to better match the harness?

hsu, 2016-02-03:
[val_gazebo pr 2](https://github.com/NASA-JSC-Robotics/val_gazebo/pull/2)

jespersmith, 2016-02-03:
Just a heads-up, we had problems with running our controller on Atlas during the VRC with high simulated joint damping. Our controller does not like it, hence we requested settable damping parameters during the VRC.

A simplified model of the actuator could be a torque source with a low-pass filter. However, our controller will work better with infinite bandwidth in simulation.

hsu, 2016-02-04:
@jespersmith Thanks for the info. For VRC, the default joint damping values for some of the Atlas joints were set conservatively high at first with lack of systems identification for damping; but it turned out the solver remained stable with a more relaxed range of joint damping values while running the *random joint torque unit test* we had implemented, thus we allowed teams to choose damping values at their own will. A better solution might be proper frequency domain based systems identification for the joints. This brings me to the next question, are there any systems id on the Valkyrie robot?

For this specific issue, the proposed [**fix**](https://github.com/NASA-JSC-Robotics/val_gazebo/pull/2) is not touching joint damping values, they remain `0.1 Nms/m` for the torso joints.

jlack1987, 2016-02-04:
We have done a significant amount of sys-id to get a good mass model, but have not done any joint friction/damping identification.

closed 2016-02-27 by jlack1987