NASA-JSC-Robotics/valkyrie #14: python workspace_checkout script doesn't automatically checkout to the val_indigo workspace

nasa-valkyrie-robot, 2016-02-29:
@pstrawse the workspace_checkout python script checks out the code to . instead of ~/val_indigo. Since we require that our code live in the ~/val_indigo directory, this script should put the code there

dustingooding, 2016-02-29:
I highly recommend we find a way to use vcs instead of this one-off script.

pstrawse, 2016-02-29:
@dustingooding There is a problem with using VCS in that it does not really support the multi-username / password thing.  I believe it was brought up with the developer, and the response was, "That's not a use-case that is covered".  When doing `vcs checkout`, the terminal gets pretty hosed.

This script is a work-around to make it easier / consistent to check out code either inside, or outside the firewall.  If people don't want to use the script, they can follow the instructions to get the workspace file and do it manually.

pstrawse, 2016-02-29:
The script simple checks out the code where the workspace file is.  If ALL the instructions on the wiki are followed, you won't have an issue with code getting checked out to the wrong place.  Hard-coding the path for the system is already a bad idea.  Making utilities like this adhere to the same structure is just worse.

kodyensley, 2016-03-01:
Hmm, we did run into issues using this. We have to setup some other systems. I'll take a closer look then.

jlack1987, 2016-03-03:
@kodyensley this one is fixed correct? Just want to confirm with you before I close it out

kodyensley, 2016-03-03:
Yeah, this is resolved.

closed 2016-03-03 by jlack1987