NASA-JSC-Robotics/valkyrie #15: uoe: yaw wandering. much worse than during training in January

mauricefallon, 2016-03-03:
The yaw estimate wanders around dramatically when the robot is standing still. By wander I mean shifting ~5 degrees left and right over about 10 seconds - directly after the execution of a whole body plan. It seems to be that the EKF compensation filter is perturbed by the motion

SCS log and some quick videos:
http://terminator.robots.inf.ed.ac.uk/public/scs/2016-03-03-yaw-wandering/

One video is from training in January and doesn't show this effect. (but there was continuous yaw drift)
http://terminator.robots.inf.ed.ac.uk/public/scs/2016-03-03-yaw-wandering/imu_not_wandering_at_nasa_2016-01-08__16-54-27.mp4

(Since training the IMU model was changed. it now has magnetometers and produces its on orientation estimate from it onboard AHRS running a compensation filter, in these cases)

jlack1987, 2016-03-03:
Just a note, there is no EKF on the IMU. We haven't been able to get very good performance out of the microstrain EKF. Usually results in her shaking around and falling over. It's currently using a complementary filter which results in the yaw drifting all over the place.

raluca-scona, 2016-03-03:
I ran the Pronto state estimator on the yaw wondering log:

http://terminator.robots.inf.ed.ac.uk/files/yaw_wondering_pronto_top.mkv

The gold robot is the IHMC estimate and the white one is the Pronto one. In the plot on the right hand side, yaw [POSE_BDI] represents the yaw from the IHMC estimate and yaw [EST_ROBOT_STATE] represents the yaw from Pronto. For this log it seems that the wandering issue is coming from the IHMC estimator rather than from the IMU.

edbot, 2016-03-10:
This should now be resolved by:
https://github.com/NASA-JSC-Robotics/val_description/pull/29

-- mfallon

closed 2016-03-14 by mauricefallon