NASA-JSC-Robotics/valkyrie #17: uoe: occasional spurious joint angle measurement

mauricefallon, 2016-03-14:
We have noticed very occasional spikes in the controller behaviour which we have now pinned down to the joint position reported from NASA containing a corrupted reading.

In this short log for a single tic the hip joint reads a joint angle of zero:
http://terminator.robots.inf.ed.ac.uk/public/scs/20160314_170257/20160314_170257_ValkyrieRosControlController_UOEPossibleMissedTickFromNASA.tar.gz

mauricefallon, 2016-03-14:
Tic 1:
[image]

Tic 2:
[image]

Tic 3:
[image]

What could be the source of this issue?

Note: that we logged the value both in IHMC's log and also in our log of the raw ROS messages from NASA.

jlack1987, 2016-03-14:
Nice catch @mauricefallon, have you caught it more than once and if you have is it always in the same joint that you see the spike?

jlack1987, 2016-03-14:
talked a bit with @jjholley and @MathisFrankB and we have a few questions. When you say you have a "log of the raw ROS messages from NASA", do you mean like a rosbag of the /joint_states topic that we publish? If so do you see it happen there also?

mauricefallon, 2016-03-14:
We have noticed it a few times before. It never causes a fall but it causes the robot to screech and sway. I'll send more logs when we observe it again. (difficult to find in the current logs)

Yes, I see it on the /joint_states topic. (Actually its /joint_states logged in an LCM log)
So I am confident that its unrelated to IHMC.

jlack1987, 2016-03-14:
ok cool, if it's seen in the /joint_states that we publish then it's definitely coming from us and not IHMC as long as the /joint_states your logging is coming from us which I assume it is. 

Next time you see this could you turn off logic power which will cause our logger to dump a log file to the NAS to /media/val_smt_logs. Could you send that to us?

jlack1987, 2016-04-01:
Alright @mauricefallon, we have finally observed this one in one of our own data logs here at JSC on another robot. This just means the issue is definitely on our side. We have a fix for it but want to do a bit more investigating to make sure we put the fix at the proper level. We'll let you know when we have pushed a fix in code.

mauricefallon, 2016-04-01:
sure, BTW, we have seen it regularly since, but haven't bothered to isolate the snippet. I assume you dont need further evidence.

jlack1987, 2016-04-01:
Yeah we have everything we need data wise. We are going to do some poking around to figure out exactly who is the culprit(synapse driver, turbodriver, API, etc), but we have everything we need and will get a fix out some time soon. Keep in mind though that we are delivering more robots soon, so it might be a few weeks before we get a chance to run all the tests we have planned.

jlack1987, 2016-04-04:
Just pushed a fix to this yesterday. It looks like the cause lies somewhere in our code for the synapse driver. We have some work going on to refactor all of our synapse code, so the true fix to this will come later. For now we have implemented a glitch filter that will remove the glitches. We did some limited testing and found that it works pretty well.

A few things to keep in mind. When a glitch happens you will miss a single tic on that particular joint's position measurement. If a glitch happens 2 tic's in a row then the filter will fail and it won't get properly filtered out. Also if the glitch happens on a joint and that joint's position reading is below the glitch filter's threshold, those also won't get filtered out. All that being said in the testing we did over the weekend we seem to successfully filter out the vast majority of all glitches that happen.

jlack1987, 2016-04-08:
Closing this one out as we have implemented a band-aid for it for now. Internal research for fixing the root cause will be ongoing though.

closed 2016-04-08 by jlack1987