NASA-JSC-Robotics/valkyrie #18: smtcore syslog error

hsu, 2016-03-15:
I am running into this issue when trying to start the val robot in sim, narrowed down

~~~
roslaunch val_gazebo val_sim_pinned.launch
~~~

to this node:

~~~
root@765125b63e71:~/ws# rosrun shared_memory_transport smtcore 
Traceback (most recent call last):
  File "/root/ws/install/lib/shared_memory_transport/smtcore", line 25, in <module>
    nasa_common_logging.configure_common_logging(level=level)
  File "/root/ws/install/lib/python2.7/dist-packages/nasa_common_logging/logging_utils.py", line 59, in configure_common_logging
    logging.config.dictConfig(config)
  File "/usr/lib/python2.7/logging/config.py", line 794, in dictConfig
    dictConfigClass(config).configure()
  File "/usr/lib/python2.7/logging/config.py", line 576, in configure
    '%r: %s' % (name, e))
ValueError: Unable to configure handler 'syslog': [Errno 2] No such file or directory
~~~

If I create a directory: `/dev/log`

the error becomes:

~~~
root@765125b63e71:~/ws/install/lib/shared_memory_transport# ./smtcore 
Traceback (most recent call last):
  File "./smtcore", line 25, in <module>
    nasa_common_logging.configure_common_logging(level=level)
  File "/root/ws/install/lib/python2.7/dist-packages/nasa_common_logging/logging_utils.py", line 59, in configure_common_logging
    logging.config.dictConfig(config)
  File "/usr/lib/python2.7/logging/config.py", line 794, in dictConfig
    dictConfigClass(config).configure()
  File "/usr/lib/python2.7/logging/config.py", line 576, in configure
    '%r: %s' % (name, e))
ValueError: Unable to configure handler 'syslog': [Errno 111] Connection refused
~~~

Any hints what I might be doing wrong?
Thanks.

dustingooding, 2016-03-15:
Do you have a syslog process running?  Preferably syslog-ng?

hsu, 2016-03-15:
How do I start the process?

~~~
ps -ef | grep syslog
~~~

returned empty for me.

dustingooding, 2016-03-15:
Assuming Ubuntu Trusty:

`sudo apt-get install syslog-ng`
`sudo start syslog-ng`

hsu, 2016-03-16:
`syslog-ng` was installed already, but I didn't have too much luck with `start syslog-ng`. In the end, I found running `syslog-ng` on commandline does indeed spawns a daemon and now smtcore doesn't crash anymore. This might be a quirk with docker (that's where I am testing things). Thanks for your help!

dustingooding, 2016-03-16:
Ahhhhh docker.  Why didn't you say so.

You need to volume mount your host's /dev/log into your container.

`docker run -it -v /dev/log:/dev/log` etc etc

That'll allow your docker process to write log files to your host's logging
system.

hsu, 2016-03-16:
got it, just starting to work with docker myself actually, thanks for the info!

dustingooding, 2016-03-17:
Let us know how it goes.  We're using docker for building CI artifacts, but
not for robot control code execution.  That's something we're working on
behind the scenes through.  I bet @kensley would be interested in this too.

kodyensley, 2016-03-17:
Yeeeelllllooo? Someone mentioned Docker?

hsu, 2016-03-21:
Hi @kodyensley Yes, we are making a docker image to test val_workspaces with gazebo7. How are you guys using docker currently? I can share with you the build script I am hacking on (currently reside in another private repository), but probably not the docker image easily since many of the repositories are still private. Send me an email and we can take this offline.

Which leads me to *the question*, are there plans to 1) make val workspace repos public and 2) release them as debian packages (bloom or ?) in the near future?

Thanks.

kodyensley, 2016-03-25:
I tried to find your email, and by tried I mean I looked for a hot minute and then life happened. Please email me at kody.g.ensley@nasa.gov and we'll take this offline. Thanks

hugomatic, 2016-04-05:
There's a new syslog related issue: 
catkin_make is failing when it is executed during a docker build command. This is because docker build cannot use syslog. 

`
[ 71%] Generating include/shared_memory_transport_sliderboard/moc_MainWindow.cpp
Traceback (most recent call last):
  File "/root/ws/devel/bin/api_builder", line 6, in <module>
    exec(fh.read())
  File "<string>", line 28, in <module>
  File "/root/ws/src/nasa_common_logging/src/nasa_common_logging/logging_utils.py", line 59, in configure_common_logging
    logging.config.dictConfig(config)
  File "/usr/lib/python2.7/logging/config.py", line 794, in dictConfig
    dictConfigClass(config).configure()
  File "/usr/lib/python2.7/logging/config.py", line 576, in configure
    '%r: %s' % (name, e))
ValueError: Unable to configure handler 'syslog': [Errno 2] No such file or directory

`
We tried starting syslog-ng inside to docker build but it is not possible:

`Step 25 : RUN service syslog-ng start
 ---> Running in 135bd0bf3da2
 * Starting system logging syslog-ng
syslog-ng: Error setting capabilities, capability management disabled; error='Operation not permitted'
Error opening file for reading; filename='/proc/kmsg', error='Operation not permitted (1)'
Error initializing message pipeline;
   ...fail!
`

dustingooding, 2016-04-06:
@hugomatic docker is only capable of running one process in a given container.  now, some people get fancy and start something like a shell, and then you can interact with that single-process shell... but when you start wanting to run services like syslog-ng, you can't really do that.

as above, start your docker container like this:

`docker run -it -v /dev/log:/dev/log etc etc`

hugomatic, 2016-04-06:
@dustingooding, thanks for pointing that out, it solves the problem when running the container (using the "docker run" command). However our problem is during build (using the "docker build" command).

Our Dockerfile runs catkin_make, and that is trying to use syslog. So catkin_make fails and the container is not built. The container was building last week, so it looks like this use of syslog during catkin_make is a new behaviour.
 
Docker build does not allow to map volume or run in privilege mode, so there is no easy way around this.

Is there way around executing devel/bin/api_builder during catkin_make? Can we add an option to api_builder  to tell it not to use the syslog handler?

It could also be that it is not a good idea to use catkin_make inside our Dockerfile. Any advice?

kodyensley, 2016-04-06:
With prior commitments I haven't been giving this thread my attention, my apologies. So what I've found for our setup, docker build is best to create a static base image. From there, start it up and run scripts/execs what have you to get the docker in the running state you want. This is how we run our CI dockers. I build a base image that has things like Indigo/xacro/few build goodies, then when we create an instance of that base image we poke the container to do things like apt-get update/rosdep update to ensure its up to date. From there we run our scripts to start the catkin_make flavor of the day. 

Docker build doesn't actually allow you to do volume mounts or set privileged mode. That is a docker run feature. So you'll need to set those things at the runtime or docker container instantiation period.

dustingooding, 2016-04-07:
The Python configuration of nasa_common_logging is a simple text file that
configures the "syslog" handler by default.  So lets try this...  Modify
nasa_common_logging/share/logging.json and remove the "syslog" element in
the root>handlers dictionary.  You can either do that prior to starting
your Dockerfile build (if you're building based on a local repo) or add a
simple 'sed' command to remove the handler programmatically prior to
catkin_make.

dustingooding, 2016-04-07:
@pstrawse would likely be interested in this, too.

hugomatic, 2016-04-09:
Unfortunately the json file is not used during catkin_make. Instead, a hardcoded common config is used.

These changes work for our use case: 
https://github.com/NASA-JSC-Robotics/nasa_common_logging/pull/1
https://github.com/NASA-JSC-Robotics/shared_memory_transport_utils/pull/1

pstrawse, 2016-04-11:
@dustingooding So this works for us because we aren't doing catkin_make inside a `docker build` command?

Looking at the PR, the only thing that gives me pause is that we will be suppressing the syslog during a build.  This might be ok since it is still logging to the screen, and that will show up in our build logs.  As long as we see api_builder log failures on-screen during a build, I'm ok with this approach.

dustingooding, 2016-04-15:
That's correct @pstrawse, it works for us because we run `catkin_make` inside an existing docker image using `docker run` that has `/dev/log` volume mounted.

I really don't like the whole `--no_syslog` flag for api_builder in https://github.com/NASA-JSC-Robotics/shared_memory_transport_utils/pull/1.  (I also don't really like the `--debug` flag, either, FWIW.)  I want there to be a way to modify a logging configuration file from the outside, so `api_builder` and all the other things we have that do logging don't all have to have these logging-specific flags attached to them.  That's why when I originally put nasa_common_logging together, both the python and c++ log configuration was done via file, so people didn't have to programmatically add flags to all their scripts to adjust effectively-hard-coded logging configurations.  So, I'd prefer the resolution to this be one that more cleanly separates the configuration of the logger from the configuration of the executable.  

For example, have a default file that gets loaded for logging configuration (nasa_common_logging/share/logging.json, or something), with a preemption load for a file in the current directory (./logging.json or something).  Make it so the scripts that do logging don't need to worry about config file names (but can certainly provide a specific config file if they want).  Make it so users of scripts can easily adjust the global logging configuration by creating a custom (even temporary) file in their current directory.  Make it so script writers can just focus on what their script is supposed to do, and not worry about how to handle log levels and handlers, and other such nonesense (to them).

hugomatic, 2016-04-15:
@dustingooding it was unfortunate of me to model  "--no-syslog" after the recent "--debug" flag. Please describe the fix on Pull Request page.

hsu, 2016-04-18:
I am curious, why is syslog functionality being invoked during build time? Can someone help me understand it in layman's terms? Thanks!

dustingooding, 2016-04-20:
We have a Python script that writes log data as a way of informing the user
of activity (as opposed to stdout).  We also have a common logging
configuration for all our Python and C++ code, and one of the default
destinations is syslog.  Put the two together and you have a script that
writes log data to syslog.  This script happens to be used during build
time to generate code/headers for other packages to use during build time.

The issue is not whether a script is writing to syslog or not.  The issue
is that there's not an easy way to stop it from doing that without
modifying the hard-coded defaults in the common logging config (and that's
the thing that should be fixed, not hard-coding yet more syslog-specific
things in scripts).  And the only reason the default common config is a
problem is because Docker doesn't have a /dev/log device available during
'docker build'.

This is certainly a problem that will affect us in the future, so it's in
our interest to fix it, too.  But we should fix it the right way.

scpeters, 2016-06-08:
This has been stagnant for a little while. Do you have any advice on what is the right way to fix it?