NASA-JSC-Robotics/valkyrie #19: Provide a method to access tare values or to apply tare to the raw signals before exposing them

mauricefallon, 2016-03-15:
Currently taring the force torque sensors is only available within the IHMC's own controller (IMO). So this makes it impossible to use the sensor properly outside of it.

Some solutions:
- NASA taring the sensors internally and only expose post-tared values via a service call.
- NASA providing a method to receive the set of tare values. (Advantage: could re-tare when walking)
- IHMC expose the tare value or tared values via their API.

What we would like is that it should be possible to access tare-ed force torque values - even if a node is launched AFTER the robot has been put on the ground.

mauricefallon, 2016-03-15:
BTW, some information about the amount of drift we might expect would be good.
(I think @jlack1987 suggested the sensors can drift after some time)

jlack1987, 2016-03-16:
This will likely make it into our next iteration, so a fix should be in for this sometime next week if not early the next.

MathisFrankB, 2016-04-01:
So we are finishing this up, here is how the new interfacing works.
-each force torque sensor has a handle (the one that is currently grabbed by IHMC) which will give
  the biased force torque values.
-each force torque sensor has an offset handle which gives the bias that is currently being applied to
  the first handle.
-There is a mode service command ('Tare') for each sensor which will tare the force and torque
  values and generate new offsets. (there is also and unTare command which reset the offset to 0)

mauricefallon, 2016-04-02:
By "biased force torque values" I assume you mean the raw measurements?

@dljsjr :
- Will IHMC use this tare interface i.e. will pressing tare in the IHMC Control UI make a call to the mode service. (If so we can access that tare value ourselves from NASA's API)
- Will biased or unbiased measurements and the tare value be available via the new ROS api

mauricefallon, 2016-04-02:
One other question: @MathisFrankB : have you had any evidence of the F/T values drifting over time?
I think Jordan suggested that they might but I didn't ask further.

So could some fancy "taring while walking" be needed?

jlack1987, 2016-04-02:
Both the tared F/T values as well as the F/T offsets used for taring are available on a ROS topic. @MathisFrankB has also provided a service call for "untaring" the sensors if you ever want to do that.

Also yes the sensors do drift a bit over time, so some fancy taring will be needed for operating for long periods of time. The pressure pads in the feet should help with this once we have them ready.

MathisFrankB, 2016-04-02:
I am in favor of making the ati reading more robust to drift. At the moment we do not have a solution for that, but it is something we can look into.

If it would be preferred to send through the raw value then let me know. When I say the tared value and the offset are being passed through, I mean that if you want to get the raw value you calculated it with
raw value = tared value + offset

dljsjr, 2016-04-04:
Sorry for the late response on this. The only blocker to us using the JSC provided interface in to the F/T sensors is that the taring/offset **feedback** data (and in the future I'm assuming also the bias/drift feedback) is not available via the ROS Control handles. Subscribing to ROS topics is not a real-time safe operation and so we can't put it in the control or state estimation loops, where we would use that information. If we can find a solution to that then we'd be happy to use the service calls to tare via the JSC API.

jlack1987, 2016-04-08:
Doug has a point. What we have implemented is not real-time safe. We can look into possibly creating a custom force/torque interface that allows you to write a boolean to signal a tare or something along those lines.

MathisFrankB, 2016-04-09:
Changing from a service call would actually be a change to the Mode Controller (I will double check if the service is threaded or not is done). The mode handle is actually exposed via a handle. The issue is that handle is defined in the jsc software. The Mode controller is currently started as part of the launch file for the controller manager. I am not against other parties grabbing these handles, but it is currently how we do things like reset actuators, and change actuator modes. Also, there is no real intelligence built in on the api level, it is all expected to be handled at the controller level. @dljsjr and @jlack1987 can let me know what you guys want to do.

jlack1987, 2016-04-15:
Going to go ahead and close this out. We are looking at reworking our mode stuff so we can offer a more real-time safe way of doing this. Until then, the service call stuff should suffice.

closed 2016-04-15 by jlack1987