NASA-JSC-Robotics/valkyrie #20: Reporting only active joints states

mauricefallon, 2016-03-23:
The robot currently has 26 active joints and with the hands on would have 52. The joint states published by NASA and IHMC don't seem to be standardised. This is a real pain when wanting to use different combinations of hand and arms.

1. Currently NASA's joint_states msg reports 52 joints
 - excluding the 4 wrist joints for some reason
 - **Request:** Only the 26 active joints should be reported. @jlack1987 : does this seem reasonable and can you make that change? (This is a blocking issue)

2. Currently IHMC's joint_states msg reports 32 joints of which 9 are static zero values
 - all except the fingers. but the IHMC controller only controls 23 joints (32 - 6 lower arms - 3 neck)
 - **Request:** IHMC should only report the 23 joints its actually controlling.

Our future usage cases are likely to be:
- our own lower arm and neck controllers paired with IHMC control.
- our own lower arm and neck controllers paired with Drake control (not using ROS messaging)

mauricefallon, 2016-03-23:
@wxmerkt  @patmarion  @symplectomorphism

wxmerkt, 2016-03-23:
https://github.com/ihmcrobotics/ihmc-open-robotics-software/issues/23 is related to (2) and (3)

jlack1987, 2016-03-23:
cc @MathisFrankB

MathisFrankB, 2016-03-23:
Messages published by the valkyrie API are more of a debug message on our part. We will soon be changing it to not actively publish at all times (unless we set a debug argument for the manager) to prevent confusion. We will additionally be changing the topic name of the published data to no longer take the special /joint_states name.

mauricefallon, 2016-03-23:
Thanks @MathisFrankB - we will avoid using those messages but they are our only source for lowlevel data outside of the IHMC controller.

In that, case its important for IHMC ( @dljsjr ) to correctly publish the joints its actually controlling. As well as other signals. 

As I understand it, its not possible for a purely passive driver/'ros_controller' to access sensing (joint states, F/T and IMU data) without blocking a controller on those joints i.e. we cannot access the joint data ourselves.

MathisFrankB, 2016-03-24:
@mauricefallon - if you would like we can add a publisher to publish joint, imu, and F/T vlaues we are getting. The issue with the current published values is where in the process it is done, and what values are being grabbed, as well as the naming of the published topics (with /joint_states being a special topic in ros).

dljsjr, 2016-03-24:
There are two issues here that pertain to IHMC:

1. Allowing outside parties to tell our controller which joints *not* to control
2. Not publishing joint data for joints we don't control

Number 2 is fairly simple if you only care about it from the IHMC ROS side of things. It gets much more complicated if you want us to exclude it from the pure Java packet API or from SCS.

Number 1 is a complicated problem. We already do this with the hands but weren't prepared for it with the neck and forearms. But we're looking in to what can be done there (see the issue that @wxmerkt mentioned). Additionally, I'm not sure how the JSC ros_control API would work with mixed-user control on arbitrary joints, because our state estimation will break if it doesn't have feedback on every joint. I'm not familiar enough with ros_control's resource management/locking to know if it's possible for multiple controllers to have read access to the joint sensors while only one has write access to the joint commands. If it can do that then we're fine. If not there's a big problem. I know that joint states are sent out over ROS but this isn't suitable for the controller; we need all the joint data atomically via the ros_control API.

We can investigate providing some sort of a pass-through API but it would probably be joint accelerations, not torques. Which shouldn't be a huge problem. But it would still take some time.

mauricefallon, 2016-03-24:
You currently have one dont you? This is just to make modifications.

We don't care about the topic names. That's easy for us to fix.

For each signal the 'what is being grabbed' is different. For the imus there is little choice except perhaps to accept a yaw bias estimate to correct for yaw drift but I think what is in place is correct. IHMC are doing a good job with that.

For the FT sensor it would be either tared or raw (for the former we'd like the tare value as we discussed in another GitHub issue). There as been some discussion about needing to tare mid experiment  e.g. While walking. To correct for tare drift.

For joint angles there is the issue with joint compliance. That's currently done only by IHMC but seems to be very beneficial we haven't looked at the difference there is between IHMC's filtering and your raw measurements. ( We will look into optimising the joint positions via vicon in April.)

That's a bunch of suggestions. Most of them are to do what the BDI api provided.

The final point I have is that ROS publishing at 500hz can't be done properly because callbacks cause messages to clump into blocks of about 50messages, IMO.
(Our control system uses LCM UDP for IPC which could achieve 1kHz)

@patmarion

> @mauricefallon - if you would like we can add a publisher to publish joint, imu, and F/T vlaues we are getting. The issue with the current published values is where in the process it is done, and [...]

The University of Edinburgh is a charitable body, registered in
Scotland, with registration number SC005336.

mauricefallon, 2016-03-24:
(Travelling this weekend)

I think this would be best resolved with a short telecon. What about next week?

We have primarily implemented neck and arm control because we haven't gotten any time frame or road map from IHMC about its implementation through the Scs controller. (Issues with neck control were first pointed out in November and again in JSC in January. ) Implementing it ourselves has been the only way for us to point the camera at the robots hands.

We are happy not to use what Wolfgang has done and give commands to Scs if we could get a proper timeframe. And relative to dynamic balance and walking this is trivial.

> 
> There are two issues here that pertain to IHMC:
> 
> Allowing outside parties to tell our controller which joints not to control
> Not publishing joint data for joints we don't control
> Number 2 is fairly simple if you only care about it from the IHMC ROS side of things. It gets much more complicated if you want us to exclude it from the pure Java packet API or from SCS.
> 
> Number 1 is a complicated problem. We already do this with the hands but weren't prepared for it with the neck and forearms. But we're looking in to what can be done there (see the issue that @wxmerkt mentioned). Additionally, I'm not sure how the JSC ros_control API would work with mixed-user control on arbitrary joints, because our state estimation will break if it doesn't have feedback on every joint. I'm not familiar enough with ros_control's resource management/locking to know if it's possible for multiple controllers to have read access to the joint sensors while only one has write access to the joint commands. If it can do that then we're fine. If not there's a big problem. I know that joint states are sent out over ROS but this isn't suitable for the controller; we need all the joint data atomically via the ros_control API.
> 
> We can investigate providing some sort of a pass-through API but it would probably be joint accelerations, not torques. Which shouldn't be a huge problem. But it would still take some time.

The University of Edinburgh is a charitable body, registered in
Scotland, with registration number SC005336.

dljsjr, 2016-03-24:
Neck control from the JSC side was still a working but still in-development feature in January. Now that it's stable, the timeline we gave @kodyensley over the phone and @wxmerkt on https://github.com/ihmcrobotics/ihmc-open-robotics-software/issues/23 was to have the neck working this week, which we are still planning on. But a lot of the "delays" come from the universal request to have us *not* release our new API and new controller architecture. Task-space head control and mixed torque-position control of the arm joints are working in the new controller architecture but this functionality cannot be back ported to the existing API, which everyone has insisted we continue to use for another month.

And so, if we release neck control this today or tomorrow (which would be possible) then control would be in joint space. You'll need to do your own IK, which I think you're doing anyway. And it will change as soon as we switch to the new API.

I'm definitely available for a telecon next week if you'd like to talk about this.

dljsjr, 2016-03-24:
To be clear, we *can* release the new API on a development branch, but the problem is splitting effort. Since we've been asked to continue to support the existing API, this has required us to invest development effort in some of the wrapper layers. We can support both APIs for a little while but it means that the "next-gen" API takes a back seat while we clean up the current-gen API and ROS packages so that it's in good working order for the robots that are getting shipped to Boston.

MathisFrankB, 2016-03-24:
Our api does not support the loading of multiple controllers trying to utilize the same claimed resource (such as controlled joints). That way we guarantee that only one controller is sending us commands.

We are just finishing implementation that allows the user to tare the force torque sensors through our api. We are also providing handles that give the tared value and the offset used to tare.

dljsjr, 2016-03-24:
@MathisFrankB I was more asking about multiple "readers", not writers, e.g. multiple controllers accessing the same JointStateInterface rather than the same JointCommandInterface. I'm not sure how you've implemented your joint interfaces though.

MathisFrankB, 2016-03-24:
yes, if they want to grab a joint state interface then they can. The joint state interface is unclaimed. Additionally it is a separate handle then the JointCommandInterface.

dljsjr, 2016-03-25:
So it sounds like the solution for toggling between us doing control and us handing over control is to switch between claiming the JointCommandInterface and the JointStateInterface. Can more than one controller access a JointStateInterface?

MathisFrankB, 2016-03-28:
I did a quick test to make sure I wasn't going to tell you the wrong thing. Multiple controllers can get the same JointStateInterface since it is not a claimed.

MathisFrankB, 2016-06-03:
@mauricefallon and @dljsjr, is there anything that needs to be done here?

mauricefallon, 2016-06-06:
We now get the joint angles directly from ROS-control and get the pelvis position from IHMC, so we are happy with the top line issue mentioned above.

closed 2016-06-06 by mauricefallon