NASA-JSC-Robotics/valkyrie #21: rttlua in controller manager and val_gazebo dies occasionally

wxmerkt, 2016-04-01:
As on Slack, controller_manager and val_gazebo alike both have rttlua occasionally dying upon loading a controller - independent on whether it's the IHMC controller or any of our own controllers. As a starting point, we've noticed that it reliably crashes when reading state or sensor handles that are not present (but we are iterating over the names that we get). Did not yet single in on a hunch/idea of why the other controllers (that would command) cause a segfault.

***ToDo:***
- [ ] send /var/nasa/common_logging as soon as the issue surfaces again

wxmerkt, 2016-04-01:
nasa_common_logging is empty/does not contain any information related to rttlua. We will wrap our getHandle in a try-catch-block for now and report back

wxmerkt, 2016-04-04:
One case is not catching unexpected packages from the IMU:
```
process[controller_spawner_link04_3214_2492486592582052579-1]: started with pid [3278]
[INFO] [WallTime: 1459775544.764566] Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1459775547.906221430]: Sensor leftTorsoImu initialized'
2016-04-04 14:12:27 processPacket  : [ERROR   ] Unsupported field in IMU packet: f1
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unsupported field in IMU packet: f1
/opt/ros/indigo/bin/rttlua: line 20:  3032 Aborted                 (core dumped) $cmd $*
[rttlua_link04_2984_169405885147980034-1] process has died [pid 3026, exit code 134, cmd /opt/ros/indigo/lib/rtt_ros/rttlua -i /home/val/val_indigo/install/share/val_deploy/scripts/startupRemote.lua -s startup_val_control -- __name:=rttlua_link04_2984_169405885147980034 __log:=/home/val/.ros/log/bfdef960-fa66-11e5-baf1-0013951a2c5c/rttlua_link04_2984_169405885147980034-1.log].
log file: /home/val/.ros/log/bfdef960-fa66-11e5-baf1-0013951a2c5c/rttlua_link04_2984_169405885147980034-1*.log
all processes on machine have died, roslaunch will exit
```

jlack1987, 2016-05-09:
@wxmerkt can this be closed out? Or any updates here? We have the IMU error captured in a different ticket and @MathisFrankB is currently working on trying to handle the bad IMU packets in a better way. Let us know.

wxmerkt, 2016-05-10:
While the IMU happens on the real robot, we do see val_gazebo crashing with some rttlua error as well occasionally (and that's what this issue should track - I posted the IMU message later when we found that, but that's more related to #34). 

I haven't been directly working with val_gazebo recently so cannot pots an updated error log, but the issue was still present as of last checking.

jlack1987, 2016-05-10:
ok i'll leave it open or now then. Let me know if you see this again.

nasa-valkyrie-robot, 2016-06-03:
@wxmerkt we have pushed a fix that will be in the next master release that should take care of the IMU issue on the real robot(see this issue). Any news on the gazebo part of this ticket?

wxmerkt, 2016-06-16:
I haven't used the most recent Gazebo release yet. I'm closing this issue for now and will reopen once I can verify that it still pops up

closed 2016-06-16 by wxmerkt

jlack1987, 2016-06-16:
Hey wolf, we have decided to pull rttlua out of the launch process and have already seen an improvement in the behavior of things because of it. Not sure if it will make the next release or not, but it is an improvement that is coming and should take care of all the rttlua issues