NASA-JSC-Robotics/valkyrie #22: Right lower arm joints have reversed axes

wxmerkt, 2016-04-02:
As reported on Slack, the joints of the right lower arm have their rotation axes reversed, cf.

[image]
https://www.youtube.com/watch?v=3HM3RhL0dJ0

We've so far seen this for:
- all pitch joints
- forearm yaw

MathisFrankB, 2016-04-06:
We have pushed a change to val description that should correct this issue.

wxmerkt, 2016-04-07:
I've just pulled from val_description, nuked build, devel, install and re-run catkin_make. Then compared the resulting URDF to the one we previously had with meld. Did not see any change apart from the midfoot_rear collision groups being added, although I am seeing your modifications. Re-run a log on the new URDF and the visualisation problem did not go away. 

Note, the resulting URDF is no different from the previous despite the changes.

jlack1987, 2016-04-07:
It's not an URDF change Wolf it's a change in the underlying finger transmissions I think, right @MathisFrankB ?

MathisFrankB, 2016-04-10:
Yes and no. The change is in the transmission for the finger joints, but the change should appear in the transmission section of the urdf. I will take a look at this Monday.

MathisFrankB, 2016-04-11:
@wxmerkt please open the following file: src/val_description/model/robots/forearm_right.xacro and see if the forearm yaw transmission reads like

<xacro:simple_forearm_position_transmission jointName="${ForearmYawActuatorName}" actuatorName="${ForearmYawActuatorName}" nodeName="${ForearmYawNode}" actuatorApi="${ForearmYawApi}" reflect="${reflect}"/>

MathisFrankB, 2016-04-11:
xacro:simple_forearm_position_transmission jointName="${ForearmYawActuatorName}" actuatorName="${ForearmYawActuatorName}" nodeName="${ForearmYawNode}" actuatorApi="${ForearmYawApi}" reflect="${reflect}"

jlack1987, 2016-04-13:
@wxmerkt what's the status here? Has this been resolved on your end?

wxmerkt, 2016-04-13:
With the robot down for a week and busy diagnosing, I didn't get a chance to test and check it yet. Will put it on the list for tomorrow (it's below getting the robot functional and available for other users though). I'll be in touch.

jlack1987, 2016-04-13:
Yeah no worries. I understand it's low priority. Just let us know whenever you get a chance. Thanks!

wxmerkt, 2016-04-15:
Fixed, thanks guys!

closed 2016-04-15 by wxmerkt