NASA-JSC-Robotics/valkyrie #23: URDF with stubs?

mauricefallon, 2016-04-04:
We see that val_description has "no hand" versions. Do these hand have stubs or do they terminate just below the elbows?

I'd like to be able to create URDFs with the stubs accurately rendered: for collision free planning and for balancing using the arms (in future)

MathisFrankB, 2016-04-04:
The no hands urdf stops at the j5 joint (which is the base of the forearm). We currently do not have a urdf with the forearm mass sims in them.

jlack1987, 2016-04-04:
I have been thinking about getting rid of the "no hands" urdf's completely and replacing it with a model that has the mass sims. Shouldn't be difficult to do.

mauricefallon, 2016-04-04:
Ok, BTW we don't use the no hands version. The models with mass sims would be useful

jlack1987, 2016-05-04:
This one is on the way. I hope you like dense, monochrome meshes :)

jlack1987, 2016-05-12:
[image]

alright @mauricefallon this one is ready to go. Pull everything and do a clean build and you should have some URDF's with the mass sims.

mauricefallon, 2016-05-13:
Thanks. Are you referring to the develop branch:
https://github.com/NASA-JSC-Robotics/val_description/tree/develop/model/meshes/arms

FYI: @MarcoCar @yimingyanged

jlack1987, 2016-05-13:
Yep that's the one

closed 2016-05-13 by jlack1987