NASA-JSC-Robotics/valkyrie #24: Force torque sensor drift causes the robot to stamp

mauricefallon, 2016-04-05:
We've regularly observed that Valkyrie begins to stamp after some time walking. I believe that it is because the F/T sensors are drifting - a very small drift of 20-30N is enough to cause the problem.
We had a similar classifier on Atlas. This is aside from https://github.com/NASA-JSC-Robotics/valkyrie/issues/19

This is an example of 3 steps where the left foot stamped in the middle:
[image]
It clearly shows that leftFootForceMag (~60N)  is above the contact threshold (50N) when in the air. Earlier in the run leftFootForceMag was about 30N in the same place.

@SylvainBertrand : is my observation correct? 
@MathisFrankB : you said you might look into this. Can I ask: 1) What model of sensor is this and 2) Is it the same model as on Atlas 3) Is there a temperature sensor on it?

Logs here:
http://terminator.robots.inf.ed.ac.uk/public/logs/20160405_083812_ValkyrieEdinburghForceTorqueDriftCausesStamping/

jlack1987, 2016-04-05:
Yeah @mauricefallon the stomping is due to drift in the six axis load cells in the feet. You can find the contact threshold force in ValkyrieWalkingControllerParameters.java. 

The sensors in Valkyrie's feet are the [ATI mini 58](http://www.ati-ia.com/products/ft/ft_models.aspx?id=Mini58) I think. Not sure what sensors Atlas used.

mauricefallon, 2016-04-05:
Thanks @jlack1987 I think this confirms my assumption. We are frustrated because this (tiny) issue sits in between NASA and IHMC and we can't fix it ourselves but it causes regular falls.

Could you discuss with @SylvainBertrand or @dljsjr and figure out a resolution?

jlack1987, 2016-04-07:
Yeah I think the best solution here is to bring in the information from the Hermes(pressure sensors on the feet) to help tare things intelligently.

mauricefallon, 2016-04-07:
Ok. By the way, with recent experiments we have seen, with the robot hanging, the force values drifting at about 1N every 30 seconds. So its seems like there could be a temperature effect.

jlack1987, 2016-04-07:
Yeah we have definitely noticed a temp effect.I usually deal with this by turning the bot on right when I get in and let them warm up while. 

I actually called ATI about this awhile back and in only a few more words they basically told us, "tough".

mauricefallon, 2016-04-11:
I have the following data point from Atlas at the DRC finals:

This shows the left/right forces (in z) at the start and end of our DRC finals day 1 run.
At the start and again 30 mins later:
[image]
Videos: [Early](https://www.youtube.com/watch?v=QoqWmrQL4o0) [Late](https://www.youtube.com/watch?v=w9SmvuCwsnI)
All the Atlas DRC runs show its possible to get stable force torque readings.

There's is a drift of 30N between start (left) and end (left) - so 1N/min in this run. I don't know if that's characteristic.
We never had problems in ground detection with Atlas but we used a 100N threshold. That's very similar to IHMC's - but IHMC use 50N for Val which doesn't leave much margin for error.

We didn't run any modelling or taring ourselves but I think (without studying it fully) that the following model could be justified:
raw_adjusted = raw + K*foot_temp
where K is a static gain estimated from testing or ATI's datasheets.

I can't make the same plots for Val as we don't see the tared values yet (https://github.com/NASA-JSC-Robotics/valkyrie/issues/19)

I don't think the approach of taring in mid stride is a great first plan. Here's some ideas:
* How sensitive is IHMC control to raising the 50N threshold - closer to 100N @dljsjr . I think this could solve the problem easily.
* How would we build a proper model for the above correction? 
* Worth checking the foot sensor model on Atlas. Perhaps the Atlas foot design had more air flow for cooling

jlack1987, 2016-04-11:
Adjusting the ATI sensor readings based on some data driven temp model would be great, but first we have to be able to sense the temperature down there. Since we don't have that sensing, it's going to be tough to compensate for it that way.

mauricefallon, 2016-04-11:
I get your point :(

 When we get back running I'll do some analysis to put a figure on the N/min drift - that might be useful