NASA-JSC-Robotics/valkyrie #25: Have ui_builder commands pick up bashrc environment settings

wxmerkt, 2016-04-14:
(As briefly discussed with Jordan) 
Do commands sent through ui_builder, e.g. "start controller manager", pick up the bash environment configuration from .bashrc/.bashrc_nasa? If not, would it be possible for these commands to pick up the same environment configuration as a SSH shell?

[Our use case is picking up a LD_PRELOAD environment configuration.]

cc @jlack1987 @kodyensley @pstrawse

dustingooding, 2016-04-14:
"start controller manager" specifically is handled with vanguard, and vanguard does source the bash environment.  you can see specifically which file is being sourced in `/etc/init/vanguard.conf` (or something close to that)  note that vanguard is not running in an "interactive" session, so not everything in `~/.bashrc` is going to work, so it's likely sourcing `~/.bashrc_nasa` (or something close to that).

if you've made a modification to a file being sourced downstream of `~/.bashrc_nasa`, you'll need to restart the vanguard service for it to see the change.  vanguard calls don't re-source environment settings with each call.

kodyensley, 2016-05-03:
Any thing else needed for this or can it be closed @wxmerkt?

wxmerkt, 2016-05-03:
We never fully resolved this and are using a workaround which also allows us to see the stdout of controller manager (quite useful). Happy to close this

jlack1987, 2016-05-03:
we're still talking about this one, lets leave it open for now

jlack1987, 2016-05-03:
I would like to try to replicate what your doing and see the behavior is at least what we expect it to be. I'll give you an example of what i'm talking about and you let me know if I need to modify it to match your use case exactly.

So in ~/.bashrc on Link I will put the line ```export TEST_VARIABLE=BLAH```. Then I will create a simple python script that does ```print os.environ['TEST_VARIABLE']``` and attach it to a button in a ui_builder UI. Then I click that button and expect to see that variable printed to the console. Maybe i'll actually log the message to syslog or something instead of print, but you get the picture. 

If the environment variables aren't getting picked up properly, then the print should throw an exception due to some non-existent environment variable or something. Does this mimic your use case? If not can you recommend changes to that example that'll duplicate what your doing?

wxmerkt, 2016-05-04:
I'm happy to give it another test and investigate early next week. Re-reading Dustin's comment I would first investigate on our end what ``.bashrc_nasa`` has in it as well as ``/etc/init/vanguard.conf`` to understand the system better. I probably made the mistake by putting it in either ``.bashrc`` or ``.bash_nasa_val.d/30_openhumanoids`` instead of the ``.bashrc_nasa``. I'll verify though to make sure and get back to you (right now we actually like seeing the stdout/ROS_INFO_STREAM from controller_manager, which otherwise won't show up in ui_builder)

wxmerkt, 2016-05-04:
Okay, so I've checked on the robot that our LD_PRELOAD is set correctly when sourcing ``.bash_nasa_val``, which is what ``/etc/init/val-vanguard.conf`` appears to be doing. Yet, when pressing the button in the user interface that variable appears to be unset. Can we perhaps test that recommended button to print as you suggested @jlack1987? Also, should all ROS_INFO and ROS_INFO_STREAM messages appear in ui_builder?

jlack1987, 2016-05-04:
Yeah we actually have some bug that's causing log messages generated in c++ to not show up in our log message window. It's kinda driving us nuts because on a few computers in our lab the c++ messages show up properly, and those computers are simple visualizer computers we setup the same as every other visualizer computer. We are actively trying to hunt that issue down. I'm not actually sure about the log messages generated by ROS. @dustingooding @pstrawse do you guys know if the ros log messages like ROS_ERROR are expected to show up in our log window?

jlack1987, 2016-05-05:
Ok just had a quick discussion with @dustingooding about how the vanguard environment(vanguard makes the remote calls for ui_builder) works, and I would like you to try something. On Link there will be a directory ~/.bash_nasa_val.d. In there there will be a number of bash files that get sourced by vanguard. If you add another bash file in there that exports the environment variable you want, then vanguard should have access to it. 

Give that a shot and let me know how it goes.

wxmerkt, 2016-05-05:
That's already how we are doing it - we've added a file in there
(50_openhumanoids) that does the export. Happy to test again Monday,
perhaps the problem is gone but didn't see due to missing logs - I'll be in
touch

jlack1987, 2016-05-31:
Alright @wxmerkt sorry it took so long but I was finally able to test this and the environment variable was echo'd as expected. Is there anything else you can think of that we can do here to try to reproduce the problems your having?

wxmerkt, 2016-05-31:
I haven't been on the robot in about three weeks. Given that we have a workaround that we prefer (i.e. so that we can actually see the log messages), we'll go with that. I believe the underlying issue did not re-occur, so I'll close this out. Thanks for your assistance :)

closed 2016-05-31 by wxmerkt