NASA-JSC-Robotics/valkyrie #27: Vicon marker mounting locations on Val

mauricefallon, 2016-04-24:
As mentioned on slack, @jlack1987 : could you give the x,y,z positions of the illustrated M6 threaded holes? We want to mount Vicon markers precisely.

[image]

I'm making an issue because we will come back to this for the torso and hands.

ruckusrobotics, 2016-04-24:
Yeah sure thing. We are in Boston this week setting up the MIT bot, but when I get back i'll get you the location to these holes.

jlack1987, 2016-04-24:
Yeah sure thing. We're in Boston this week setting up the MIT bot, but i'll get you the coords of these holes when I get back.

jlack1987, 2016-05-02:
Here is the same image with the cartesian position of those points marked. I marked the two on the pelvis bone on the robots right side, but didn't mark the left because they are just mirrored, so you can just negate the y-component of the ones on the left to get the ones on the right. Let me know if ya need any more!

[image]

closed 2016-05-03 by jlack1987

christian-rauch, 2016-07-18:
We now would like to track the left hand pose with the Vicon system and intent to attach markers to the threaded M3 holes that hold the back plate.
[image]
@jlack1987 can you provide use with the 3D positions of these screws with respect to the "**leftPalm**" frame?

jlack1987, 2016-07-19:
[image]

Here ya go. Note that I flipped the image over.