NASA-JSC-Robotics/valkyrie #28: Thumb roll actuator command

gizatt, 2016-04-29:
Issuing command to hardware_interface::JointHandle acquired from hardware_interface::PositionJointInterface for the thumb roll on our arm testbench ultimately results in a .../thumb_roll/desiredPodRad value (visible on smtdashboard) that is offset by an amount.

That is, when I issue a command through the roscontrol API for a thumb roll position of 0.5, the SMT desired position goes to 2.24. When I issue a command for 1.5, it goes to 3.24. If I manually command a value between ~0.5 and ~1.5, the thumb moves normally, but it doesn't move when commanded to these out-of-range values through the roscontrol interface. 

Also, for what it's worth, .../thumb_roll/Mot_tempC reports -32768.

I can probably account for this manually in my driver, but this feels like a calibration issue. Any advice?

wxmerkt, 2016-04-29:
Have you checked that the actuator serial numbers are matching the stickers on the hand in the testbench_....xml file?

gizatt, 2016-04-29:
Yeah, they check out and I rebuilt to be sure.

wxmerkt, 2016-04-29:
We also had to nuke the ``build``, ``devel``, ``install`` directories before re-building because xacro didn't always replace them correctly in ``install``

gizatt, 2016-04-29:
I nuked everything too!

jlack1987, 2016-04-30:
Could you create a log file of this and send it our way when you get a chance? Since this is on a testbench, you'll need to start the logger manually before you start running your controller. Before you get started, just run the command,

```smtlogger -r 500```

then proceed normally with running your controller. Once your done just Ctrl+C the logger and it'll dump a log file in place. If you could send that to us we'll take a look at it. Thanks!

cc @MathisFrankB

MathisFrankB, 2016-05-02:
There is no temperature sensing on the thumb roll motor. The -32768 is a saturated value. The values that you can command the thumb roll to depend on the arm being used. Please specify which forearm that you are using so we can investigate this on our testbench (right or left).

gizatt, 2016-05-02:
It's a left forearm, serials v_g_007 and v_h_003. I'll edit this to link to the log in a minute

In the log, I: roslaunch the val_controller_manager, load in my command plugin, cfgLoader and streamFunctions in the configuration info, reset->park->positionMode both athena boards, and then issue a series of commands to the thumb roll to try to make it move back and forth. I also issue some commands to the other fingers to open and close after that, in case that's useful. FWIW, all of this was done with motor power off.

jlack1987, 2016-05-23:
Hey Greg just wanting to say we haven't forgotten about you. We actually don't have an assembled, working forearm to test with right now. As soon as we do we'll address this and get back to you. Sorry for the dalay

MathisFrankB, 2016-06-02:
We have looked into this and have found the issue. There is a hot fix that is in review that will fix the problem.

jlack1987, 2016-06-03:
@gizatt a hotfix for this has been merged into master. Please let us know if you continue having problems

gizatt, 2016-06-03:
Thanks! I'll give this a whirl as soon as I can.

nasa-valkyrie-robot, 2016-06-20:
@gizatt any updates here? Let us know

gizatt, 2016-06-24:
Looks like it's behaving normally now. Thanks!

closed 2016-06-24 by gizatt