NASA-JSC-Robotics/valkyrie #29: rosdep error: libflycapture2

tkoolen, 2016-04-29:
Hey, I'm trying to set up the Valkyrie software stack according to https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Setup-Valkyrie-Developer-Environment and https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Valkyrie-Source-Code. On the rosdep install step I'm getting:

```
twan@tk-sloth:~/val_indigo$ rosdep install --from-paths src -i -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
flycapture2_gui_pkgs: Cannot locate rosdep definition for [libflycapture2-gui]
flycapture2_pkgs: Cannot locate rosdep definition for [libflycapture2]
```

Any ideas?

tkoolen, 2016-04-29:
I was able to work around this issue by deleting val_vision.

jlack1987, 2016-04-30:
Yeah the val_vision is a package we had to create for the belly cams on Val due to incomplete install instructions from the PGR folks. Your not the first person to run into this, and I think the issue is due to the fact that we aren't allowed yet to give external access to our debians. Correct step is to just remove val_vision for now. Anything I left out @kodyensley ?

oh-dev, 2016-04-30:
FYI, We have LCM drivers for these cameras in the OpenHumanoids stack. 

I think an issue with Point Grey cameras is that their API is proprietary. (Same as the fish eyes on Atlas)

tkoolen, 2016-05-02:
OK, thanks for the info.

closed 2016-05-02 by jlack1987