NASA-JSC-Robotics/valkyrie #30: Missing dependency: libffi-dev

tkoolen, 2016-04-29:
During the rosdep install step in https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Valkyrie-Source-Code, I got
```
    No package 'libffi' found
    c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
```

Fixed by running `sudo apt-get install libffi-dev`

jlack1987, 2016-04-30:
hmm I have never heard of that honestly, @kodyensley does this look familiar to you at all?

jlack1987, 2016-05-01:
Ok just an update on this, it looks like we have some unit tests that are failing due to this same issue, so we appear to have had a dependency slip through the cracks somewhere. Sorry about this, we'll get it updated as soon as possible.

tkoolen, 2016-05-02:
No problem. Install process was pretty smooth despite these few minor issues, so kudos.

jlack1987, 2016-05-02:
Thanks glad to hear. All credit goes to @kodyensley for that.

@kodyensley ++ for awesome configuration management skills

kodyensley, 2016-05-03:
Hey guys, 

I traced this down to a new version of python-paramiko. It seems it was built against the libffi-dev package and not the libffi6 package. I've amended our install tree to pull this in automagically for now but I'm adding a standard disclaimer: "Not our fault". 

The instructions on the github wiki should be amended 'soon'. I'm just trying to test all the edge cases to ensure something else didn't popup. 

Thanks for fighting through this @tkoolen!

nasa-valkyrie-robot, 2016-05-09:
We have temporarily added libffi to our dependency tree to resolve this issue. Let us know if you have any more issues.

closed 2016-05-09 by nasa-valkyrie-robot