NASA-JSC-Robotics/valkyrie #31: Can't install visualizer

tkoolen, 2016-05-02:
While following [this guide](https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Get-Valkyrie-Code), I tried installing the "MIT Vis02" visualizer, but I got this:

[image]

Any ideas?

jlack1987, 2016-05-02:
You tried installing this on your personal computer or what? The 'MIT Vis02' is specifically for your visualizer computer that's running the robot and is already installed. Also there is an actual link on that page for the debian 'MIT Vis02' that you want. Unfortunately we are not allowed to make our debs public, so we aren't able to make it all sudo apt-getable.

tkoolen, 2016-05-02:
> You tried installing this on your personal computer or what?

Yup, I'm just a robot following instructions to the letter and this looked like it needed to be done ;-)

closed 2016-05-02 by tkoolen

jlack1987, 2016-05-02:
Did you download the debian [here](https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Setup-Valkyrie-Developer-Environment). It's the debian under the 'Generic System Config' link

tkoolen, 2016-05-02:
Yeah, I downloaded the generic system config one. I really just followed instructions word for word. I closed the issue because clearly I wasn't supposed to do the MIT vis step anyway.

jlack1987, 2016-05-02:
Ok I think  I see the issue. It's not so clear that unless your installing a visualizer computer, you don't need to continue after checking out our code. I have put a clarifying statement in the wiki that will hopefully prevent others from making the same mistake.