NASA-JSC-Robotics/valkyrie #32: roswtf issues

tkoolen, 2016-05-03:
After a clean install I ran `roswtf` to try to debug a val gazebo sim issue. It reported some issues (that were probably not related to the gazebo sim).

Missing dependencies:
* roslang (resolved after `apt-get install ros-indigo-roslang`)
* angles (resolved after `apt-get install ros-indigo-angles`)

I still have the following error:
```
ERROR Not all paths in ROS_PACKAGE_PATH [/home/twan/val_indigo/install/share:/home/twan/val_indigo/install/stacks:/opt/nasa/indigo/share:/opt/nasa/indigo/stacks:/opt/ros/indigo/share:/opt/ros/indigo/stacks] point to an existing directory: 
 * /opt/nasa/indigo/share
```

hsu, 2016-05-03:
Is this a roswtf bug? This message seems to say `/opt/nasa/indigo/share` does not exist or is not a directory on your computer, is that the case or is roswtf confused?

tkoolen, 2016-05-03:
That directory does not exist on my computer. Could it be something that snuck in on a JSC computer? 

This isn't high priority by the way; as far as I know it's not causing anything to malfunction.

jlack1987, 2016-05-03:
That directory should exist on your computer. I think it's created when you execute,

```sudo dpkg -i nasa-indigo-workspace*.deb nasa-val-system-config*.deb python-pyqtgraph*.deb```

Do you recall if that command was successful?

tkoolen, 2016-05-03:
I believe those commands were successful. I'll try to rerun them. `/opt/nasa/indigo` exists, but it doesn't contain a `share` folder (only `env.sh  etc  setup.bash  setup.sh  _setup_util.py  setup.zsh`)

tkoolen, 2016-05-03:
reran that command, completed without error, `/opt/nasa/indigo/share` still doesn't exist.

jlack1987, 2016-05-03:
hmm ok, might need to bring in the big guns. cc @kodyensley

kodyensley, 2016-05-03:
As to the missing debs listed by @tkoolen, I believe those may be an issue with us removing the 'ros-indigo-ros-base' dependency from our install tree. None of 'our stuff' requires that to be installed so we removed it. Seems like there are more than a few ROS things that don't properly hunt down their own deps? We've added ros-indigo-ros-base back into our install instructions for now. We'll revisit pairing down our dep tree in the future. 

As to the /opt/nasa/indigo/share directory not existing, if you have none of our Debian's (other than nasa-indigo-workspace*.deb nasa-val-system-config*.deb python-pyqtgraph*.deb) installed, which you currently do not, the share directory won't exist. 

Not sure why thats tossing an error, thats new to me. We've been doing things this way for a few years and haven't seen an issue... though we don't use roswtf.

kodyensley, 2016-05-03:
Yeah I can confirm @tkoolen it looks like roswtf does call that out as an error. It frankly won't hurt you, but its something we can slate to look into on our side. Also that error will go away when we switch you guys from source code to debs. 

Thanks for the catch.