NASA-JSC-Robotics/valkyrie #33: Gazebo sim errors related to libati_sensor_interface.so

tkoolen, 2016-05-03:
I'm having issues with the Gazebo sim. Everything _seemed_ to be working OK yesterday when I installed everything (followed https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Simulating-Valkyrie, both the floating base sim and the pinned sim ran), but now I get errors in the console and Gazebo only shows a black screen.

Command:
```
roslaunch val_gazebo val_sim.launch
```

Notable console messages:
```
/opt/ros/indigo/bin/rttlua-gnulinux: symbol lookup error: /home/twan/val_indigo/install/lib//libati_sensor_interface.so: undefined symbol: _ZN4YAML6detail9node_data14convert_to_mapEN5boost10shared_ptrINS0_13memory_holderEEE
[gazebo-5] process has died [pid 9068, exit code 127, cmd /opt/ros/indigo/lib/gazebo_ros/gzserver -u -e ode /home/twan/val_indigo/install/share/val_gazebo/worlds/empty.world __name:=gazebo __log:=/home/twan/.ros/log/79efbc7e-1176-11e6-833b-0000000002f7/gazebo-5.log].
log file: /home/twan/.ros/log/79efbc7e-1176-11e6-833b-0000000002f7/gazebo-5*.log
[rttlua_tk_sloth_9258_7795417012112573363-1] process has died [pid 9285, exit code 127, cmd /opt/ros/indigo/lib/rtt_ros/rttlua -i /home/twan/val_indigo/install/share/val_deploy/scripts/startupRemote.lua -s startup_val_control -- __name:=rttlua_tk_sloth_9258_7795417012112573363 __log:=/home/twan/.ros/log/79efbc7e-1176-11e6-833b-0000000002f7/rttlua_tk_sloth_9258_7795417012112573363-1.log].
log file: /home/twan/.ros/log/79efbc7e-1176-11e6-833b-0000000002f7/rttlua_tk_sloth_9258_7795417012112573363-1*.log
```

so that seems to be pointing to libati_sensor_interface.so.

The log file `/home/twan/.ros/log/79efbc7e-1176-11e6-833b-0000000002f7/rttlua_tk_sloth_9258_7795417012112573363-1.log` does not exist.

What Gazebo looks like after startup:
[image]

Note that one time today, seemingly randomly, the Valkyrie 3D model did show up in Gazebo and I was able to simulate it, but I believe the rttlua error message was still present. I'm not 100% sure whether that rttlua message was present yesterday when I was able to simulate.

Any ideas?

jlack1987, 2016-05-03:
This has actually happened to me a few times today and yesterday. It only happens once in awhile so I have kinda just been ignoring it

tkoolen, 2016-05-03:
which of the (possibly separate, possibly connected) issues have you encountered? The libati_sensor_interface.so undefined symbol errors, or the black screen in Gazebo?

tkoolen, 2016-05-03:
Digging around a bit, the undefined symbol error seems to be a yaml-cpp thing
https://stackoverflow.com/questions/36403287/compiler-error-with-yaml-cpp-undefined-reference-to-yamldetailnode-data

tkoolen, 2016-05-03:
OK, the undefined symbol error turns out to be our issue. `ldd libati_sensor_interface.so` has a line
```
libyaml-cpp.so.0.5 => /home/twan/code/oh-distro/software/build/lib/libyaml-cpp.so.0.5 (0x00007f5156d3f000)
```
so in **our** software stack. It's caused by a bad interaction (PATH modification) between a script that we source and the Valkyrie install.

I can simulate again without errors after doing a
```
catkin_make clean
catkin_make install
```
in `~/val_indigo` in a terminal where that path modification was not done.

I still get the black screen every once in a while, but not (almost?) always like before.

Sorry about the noise.

closed 2016-05-03 by tkoolen

jlack1987, 2016-05-04:
No worries, glad ya got it figured out. Let us know if you have any other issue!