NASA-JSC-Robotics/valkyrie #34: control_manager fails to find IMU

manuelli, 2016-05-08:
About half the time we power on the robot and launch the control_manager we get an error related to a bad connection to the IMU.

> 2016-05-08 16:03:51 processPacket  : [ERROR   ] Unsupported field in IMU packet: f1
2016-05-08 16:03:51 callbackFilter : [ERROR   ] Imu either unplugger or has bad connection
7.071 [ ERROR  ][/opt/ros/indigo/bin/rttlua-gnulinux::main()] in updateHook(): switching to exception state because of unhandled exception
7.071 [ ERROR  ][/opt/ros/indigo/bin/rttlua-gnulinux::main()]   Imu either unplugger or has bad connection

Sometimes just relaunching the control_manager fixes it but other times it requires power cycling the robot.

MathisFrankB, 2016-05-08:
We are looking into this, but it appears to be a problem in the imu itself (possibly the power two the imu). That exception is thrown after receiving one full packet from the imu, and the checksum is verified. Certain bytes of the packet are checked against possible good values, and if they do not match then an exception is thrown. It is unclear though why exactly the imu is sending the bad data through (which is why the exception). Until we know more, you can check that the imu cable is firmly seated in the computer board (I will see where you might be able to put some strain relief on those cables). You might also try switching to a different usb port.

mauricefallon, 2016-05-09:
We've seen the same problem in Edinburgh

MathisFrankB, 2016-06-02:
We have looked into this and have added some checking that help alleviate the issue. This will be available to the master branch in the next release.