NASA-JSC-Robotics/valkyrie #35: Which version of gazebo to use for val

scpeters, 2016-06-10:
Val is currently using trusty, ros indigo, and gazebo4 (similar to the drcsim package). There is a [discussion thread](https://bitbucket.org/osrf/gazebo/pull-requests/2316/issue-1959-imu-fixes/diff#comment-19623533) about switching to a new version of gazebo for val that can be continued here.

@hsu suggested that we upgrade val to use gazebo7.

@dljsjr placed the following requirements:

1. it can run atlas. (there is a thread about releasing a new drcsim that uses gazebo7 [here](https://bitbucket.org/osrf/drcsim/issues/497/will-drcsim-debs-be-released-for-gazebo5))

2. it works with ROS Indigo. (that is no problem, we have debs called `ros-indigo-gazebo7-ros-pkgs` for trusty)

3. it should be installable via `ros-<release>-desktop-full`. (this would not be easy)

cc: @dljsjr @hsu @jlack1987 @j-rivero

dljsjr, 2016-06-10:
The reason I included 3 is because the current JSC installation instructions currently leverage `ros-indigo-desktop-full` in their setup instructions, and then installing Gazebo 4 after the fact. It's brittle and error prone. So they currently require all installations to be done on a computer that doesn't already have Gazebo, or their instructions will fail.

I suppose the other option is for JSC to isolate exactly what ROS packages that they use but I imagine that'll be extremely tricky.

jlack1987, 2016-06-10:
cc @dustingooding @kodyensley

scpeters, 2016-06-10:
It wouldn't fix the problem you describe, but we can maintain the status quo but switch from gazebo4 to gazebo7 by creating a rosdep rule similar to [this one](https://github.com/NASA-JSC-Robotics/nasa_common_rosdep/blob/master/nasa-gazebo4.yaml).

dustingooding, 2016-06-11:
I've confirmed that ros-indigo-desktop (not full) and the custom rosdep
keys work fine together (on R2), and all the "missing" things in
not-full-desktop get installed correctly given correct dependencies listed
in package.xml files (with the use of rosdep/Debian dependencies).  Val's
use of desktop-full is a dangerous crutch and should probably be gotten rid
of... ideally you'd only need ros-indigo-ros-base and rely on dependency
trees for the rest.

kodyensley, 2016-06-13:
All, 

We removed our dependency on ros-indigo-desktop-full about two months ago.

jlack1987, 2016-06-13:
[val_gazebo](https://github.com/NASA-JSC-Robotics/val_gazebo/tree/feature/VDEV-1212) branch setup for Gazebo7. Gazebo integrated unit tests passing.

closed 2016-06-20 by nasa-valkyrie-robot