NASA-JSC-Robotics/valkyrie #36: Forearm Yaw running off upon servoing lower arm

wxmerkt, 2016-06-22:
As discussed on Slack, the forearm yaw runs off relatively frequently (today 4 out of 5 times).

Logs are on Slack, and here's the plot of desired/commanded vs actual, note how it is running off away from the desired command.

[image]

We'll add more information once we are reproducing this issue.

jlack1987, 2016-06-24:
@wxmerkt any new info here?

wxmerkt, 2016-06-25:
I'm pretty busy at the moment so not on the robot. Unless someone plans on doing experiments using the lower arm, we'll likely won't use and servo the forearm before upgrading. I'll add more information and ping you guys on Slack once we see this again