NASA-JSC-Robotics/valkyrie #37: uoe: persistent lean introduced in March

mauricefallon, 2016-06-23:
We've been seeing our robot leaning onto the left foot consistently - ever since March during installation. We never sought to fix it until realising how characteristic it is and that the robot usually falls to the left when walking also.

Looking through the logs, its clear that up to March 02 (including testing of the robot in JSC), the robot didn't lean at all. Measures visualised in LogVisualizer are symmetric just after standing the robot:
[image]

For the next run on March 03 we see the robot leaning onto the left foot (again just after standing the robot). All metrics are shifted to the left except the Desired Capture Point:

[image]

Here are two SCS log snippets, one from before and the other after:
* [March 02 - before left lean introduced](http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160302_185729_ValkyrieRosControlController_SnippetBeforeLeftLean.tar.gz)
* [March 03 - after left lean introduced](http://terminator.robots.inf.ed.ac.uk/bignas/useful/ihmc/20160304_103814_ValkyrieRosControlController_SnippetLeftLean.tar.gz)

@MathisFrankB : My first guess was that the IMU was replaced then but I **DONT THINK* that's to blame. We didn't replace it until a week later.

@jlack1987 : What do you think could have caused this. My first guess is that we might have redone the calibrations during the installation. Where are these calibration offsets stored?

I'm hoping to figure this one out before changing the controller and using the new calibration figures.

jlack1987, 2016-06-23:
The calibration is stored in xml files for each actuator. If you look in val_description/instance/coefficients/actuators you'll see a bunch of v_x_xxx.xml files. A coeff named 'PositionOffset_Rad' will be in those files and those represent a position offset applied to the encoder value for zero. To see the serial numbers for your robot you can either look in the instance file(~/val_indigo/install/share/val_description/instance/instances/instance_files/valkyrie_A.xml) or in ~/val_indigo/src/val_description/common/xacro/serial_numbers/valkyrie_A_serials.xacro.

mauricefallon, 2016-06-23:
The git history says that new calibration was done between these experiments:
https://github.com/NASA-JSC-Robotics/val_description/commit/da9f4c45650fc8207ae98b03f262440bf2476b7e
And commited at 2016-03-02 16:59:28

Your guys stuff is pretty solid: that I back this out by a combination of 3 month old logs and git history is pretty cool.

Before we look at reverting at least some of these calibrations, can I ask:
* how did you achieve such good ICP tracking in those original experiments?
* do you think that the combined set of calibrations is bad or would just changing the legs back improve i.e. can bad arm calibrations effect ICP tracking.
* And assuming the calibrations are to blame, why do you think we did this so badly? (I see some arm joints changed quite a bit one of the ankles did change by over a degree, which is significant)

jlack1987, 2016-06-23:
Yeah bad arm calibrations can result in bad ICP tracking since it'll result in a bad overall COM estimation which the ICP is a function of. Was this the calibration we did while we were there or did you guys do this one yourselves? 

I'm really interested to see if reverting these calibrations fixes your issue. So if you don't mind trying that I would appreciate it even if you revert right back. If it fixes your problem then I guess we need to revisit the design of our calibration fixtures and/or our calibration procedure.

Pretty much every robot has been biased to one side a little bit(usually a really small amount), but looking at the measured COP it's noticeable. We have spent tons of time going over logs looking for the cause, and we have even questioned calibration, but it's difficult to prove that that small calibration errors are causing it to stand off center especially with a shitty IMU that we are constantly pointing the finger at.

Let me know what you think.

mauricefallon, 2016-06-23:
> Was this the calibration we did while we were there or did you guys do this one yourselves?

The one when you where here. We haven't changed them since. It's really noticeable visually. And the left foot F/T usually has about 70% of the mass on it. 

We will run the experiment in the next few days.

We could really help with analysing the IMU performance: our Vicon system makes this so easy. If my recollection is correct pitch and roll estimate IHMC was was much better than I expected.

edbot, 2016-06-24:
( @mauricefallon  here )

I tried reverting the March calibrations but it didn't improve the biased stance.

I changed the PositionOffset_Rad values for each of the joints (in the v_x_xxx.xml files) and made sure they were installed to:
`val_indigo/install/share/val_description/instance/coefficients/actuators`
On both link04 and vis04.

Is there anything else that I need to do to push the changes to the robot itself?

(I also introduced a fake knee offset for sanity checking - which I could see in SCS. So I think I am doing things properly)

For posterity these are the revisions I used
* revision containing the March calibrations: da9f4c45650fc8207ae98b03f262440bf2476b7e
* rev just before the calibrations: 03e37b71f07c5873da65464fbe58ecf6ad406123
* rev on vis04 and link this morning: 996d8abed9d50a49693dae7486111c6ce01d1a9e
  * Commit: "Fixed spelling again. Sorry again wolf."
* Branch which reverts the calibrations: https://github.com/NASA-JSC-Robotics/openhumanoids:mf-revert-installation-calibration

jlack1987, 2016-06-24:
@mauricefallon since you verified they were in the install tree then as long as you loaded coeffs again then they should have been applied. A way to **guarantee** that they are applied is to have an actuator stream its coeffs back to you and you can check the 'PositionOffset_Rad' value to make sure it's what you think it is. To do this, from a terminal on the bot, do

turboStream -c <channel> -n <node> config

where <channel> can be left_leg or left_arm or whatever channel the actuator lives on and node is like j3 or j4 for the particular node your looking for. It will then print in the terminal a list of coeffs that are on that actuator. It will also fault the joint, so keep that in mind and don't do it on the knee or something while she is standing. A complete example if I want to check the left hip pitch's coeffs I would execute,

turboStream -c left_leg -n j3 config

Let me know if you have any issues with this.

MathisFrankB, 2016-06-29:
@mauricefallon I am just going back through this. Did the robot fall in between the two plots? It is a little out there, but it might be possible that something got knocked loose during a fall. It is also possible for a large fall to cause the imu to loose calibration. This would be pretty far fetch since the imu is supposed to be able to handle up to 50 gs before it loses calibration, but with how far off some of their other specs have been it would surprise me.