NASA-JSC-Robotics/valkyrie #38: Launching cfgLoader for forearm testbench errors out

gizatt, 2016-06-24:
After a clean rebuild on the most recent develop branch for all Val repos, when I try to start up the forearm testbench, ```smtcore``` and ```control -r ...``` both work fine, but when running ```cfgLoader -i .../forearm_testbench_left.xml```, I get the error:

```
Traceback (most recent call last):
  File "/home/val/val_indigo/install/bin/cfgLoader", line 255, in <module>
    targets, cfg = determineCfgs(args, logger)
  File "/home/val/val_indigo/install/bin/cfgLoader", line 204, in determineCfgs
    cfg = InstanceFileHandler.InstanceFileHandler(args.instance)
  File "/home/val/val_indigo/install/lib/python2.7/dist-packages/val_description/InstanceFileHandler.py", line 44, in __init__
    self.processInstanceFile()
  File "/home/val/val_indigo/install/lib/python2.7/dist-packages/val_description/InstanceFileHandler.py", line 64, in processInstanceFile
    self.turboPublicProcSources = {'firmware': {'type': 'url', 'location': turboProcFirmwareRoot.find('PublicProcFirmwareBaseURL').get('id')}}
AttributeError: 'NoneType' object has no attribute 'find'
```

Looks like it's looking for a field (turboProcFirmwareRoot) not included in that XML file. Any tips?

jlack1987, 2016-06-24:
@gizatt this is some newly added stuff, we'll look into it and get back to you asap

jlack1987, 2016-07-06:
@gizatt has this been taken care of yet? I feel like I remember you working with @jjholley to get this fixed?

gizatt, 2016-07-06:
Yup, this has been resolved and my testbench is happily up and running. Thanks!

closed 2016-07-06 by gizatt