NASA-JSC-Robotics/valkyrie #39: Rosdep Sources for Gazebo 7

velinddimitrov, 2016-07-05:
Rosdep sources from install instructions should be updated for Gazebo 7.

sudo sh -c 'echo "yaml https://raw.githubusercontent.com/NASA-JSC-Robotics/nasa_common_rosdep/master/nasa-gazebo4.yaml" > /etc/ros/rosdep/sources.list.d/12-nasa-gazebo4.list'

should read

sudo sh -c 'echo "yaml https://raw.githubusercontent.com/NASA-JSC-Robotics/nasa_common_rosdep/master/nasa-indigo-gazebo7.yaml" > /etc/ros/rosdep/sources.list.d/12-nasa-gazebo7.list'

velinddimitrov, 2016-07-05:
@kodyensley

scpeters, 2016-07-05:
Are you referring to this wiki page?

* https://github.com/NASA-JSC-Robotics/valkyrie/wiki/Setup-Valkyrie-Developer-Environment

velinddimitrov, 2016-07-05:
Yes

kodyensley, 2016-07-05:
This is sort-of expected. We pushed some things a bit early and they require gazebo7 keys. The documentation didn't catch up. Thats my bad. I'll have that fixed today.

kodyensley, 2016-07-05:
This should be fixed now. Note that because the rosdeps share the same name the old key needs to be removed or rosdep will resolve first come/first serve. (The install instructions also have this added).

    sudo rm /etc/ros/rosdep/sources.list.d/12-nasa-gazebo4.list

closed 2016-07-05 by kodyensley