NASA-JSC-Robotics/valkyrie #40: accessing the force torque tares

mauricefallon, 2016-07-14:
From a recent run, the force torque offsets are still not available to us: i.e. the new IHMC controller still publishes the untared values.

@MathisFrankB : is the tare mode in the Cartman release?  (apologies, but i don't know when this was fixed) Is there a hardware interface which stores these tares? 

And do you know if IHMC write their tares to that interface? (so that we can access them)

jlack1987, 2016-07-14:
Yeah @mauricefallon I just double checked and it looks like that particular fix isn't in Cartman. The way we have it working is there is a mode interface for the force/torque sensors that allows you to call tare/untare, and we have stopped claiming those resources, so they are available for anybody else to claim. In the Cartman release though, it looks like we're still claiming them. In the Daria release, we are no longer claiming those resources so you or IHMC(cc @dljsjr) can claim them and tare those sensors directly via the mode handle. cc @MathisFrankB

mauricefallon, 2016-07-14:
Thanks, that's good to know.

@dljsjr : would it be possible for you to push out the tared values ( leftFootStateEstimatorForceEstimatorForceZ) to ROS 
instead of the raw values (raw_force_z_leftAnkleRoll)?

closed 2016-07-14 by mauricefallon